raspberrypi/userland #654: at jobs queued before shutdown do not run to schedule

ajuniper, 2020-10-31:
If I create an at job and then shut the system down, the job will run late if the system is powered on before the scheduled time.  The longer the system is shut down, the larger the delay will be.

Steps to reproduce:

1. pi is booted, on wifi and ntp is synchronised
1. create an at job: `at now+1h <<<'logger "this job should run 1 hour after $(date)"`
1. shut the pi down
1. wait 30 minutes
1. power the pi on - it joins the wifi and sync's with NTP within a minute
1. wait for the at job to run and observe the time at which it runs

I've observed jobs running 10-25 minutes late, with the scheduled time occurring >30 after the pi is powered on.

Gut feel is that atd records how far into the future existing jobs must run before ntp is synchronised and that it is not aware that time has jumped when ntp synchronisation occurs.

at jobs scheduled after the power on run as expected.

ajuniper, 2020-10-31:
Apologies, just realised this was entirely the wrong place to open this ticket, will find somewhere better....

closed 2020-10-31 by ajuniper

HinTak, 2020-10-31:
I think you should file at raspbian /debian against the atd package.