raspberrypi/userland #656: raspivid -3d tb is working only at limited resolutions

wkeeling63, 2020-11-09:
Issue:
raspivid -3d tb only works at the following resolutions (480X864, 1040X1824, 1056X1856, 1080X1920, 1088X1920, 1112X1920, 1120X1920, 1144X1920 and 1152X1920)

Command:
raspivid -3d tb -w 1280 -h 720 -t 5000 -o 720X1280.h264
or any resolution other those listed above.

Correct behavior:
Encoded video at all resolutions up to width of 1920. I have read 6by9’s rules for stereo mode but I am not sure what the resolution limits for top/bottom are.

Actual behavior:
tb only works at the following resolutions (480X864, 1040X1824, 1056X1856, 1080X1920, 1088X1920, 1112X1920, 1120X1920, 1144X1920 and 1152X1920).

Environment:
Compute Module 3+

Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage5

Apr 27 2020 14:36:33 
Copyright (c) 2012 Broadcom
version 3a8f1793b758d6fb7f375edaa260e069ecd34c88 (clean) (release) (start_x) 

Linux pi2cam 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

6by9, 2020-11-10:
```
Apr 27 2020 14:36:33
Copyright (c) 2012 Broadcom
version 3a8f1793b758d6fb7f375edaa260e069ecd34c88 (clean) (release) (start_x)

Linux pi2cam 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
```
Update your system. This was fixed in July - https://github.com/raspberrypi/firmware/commit/542aceb30364db3ce87d532ea90451f46bbb84e9 "firmware: IL: camera: Fix stereoscopic pool allocations"

wkeeling63, 2020-11-10:
thanks you ma'am/sir :),

update to:
pi@pi2cam:~ $ vcgencmd version
Nov  6 2020 18:01:30 
Copyright (c) 2012 Broadcom
version 9ae94aeb1241adffec586ae451862e3fa6e05e6a (clean) (release) (start_x)

Verified at 480x864, 720X1280 and 1080X1920

closed 2020-11-10 by wkeeling63