raspberrypi/userland #671: raspistill: feature request: cross-hair for the viewfinder

afkrause, 2020-12-29:
Simple feature request: would it be possible to add a simple cross-hair to the view-finder for raspistill?

that would help to center an object exactly into the middle of the raspicam.

6by9, 2020-12-29:
Not easily, no. It would need to be added as an additional display layer over the top of the preview, and whilst that's possible, you then have to mess around with where the --win option has put the preview window, how aspect ratio changes could move the image, etc. 

Far easier to get a small sticker and put it on your display, or a piece of sellotape that you mark your centre point with.

afkrause, 2020-12-29:
thank you very much. sometimes the easy and obvious solution is hard to see! i will stark raspistill in fullscreen mode and place a sticker right to the centre of my monitor. that should to the trick!

6by9, 2021-03-19:
Issue resolved so closing. Please reopen if you feel there is anything further required.

closed 2021-03-19 by 6by9