raspberrypi/userland #672: how to match output with input in mmal

linhuifj, 2020-12-30:
When the input rate is larger than the decoding rate, will the packet be dropped? How can I know which input buffer is decoded when I get an output buffer?

6by9, 2020-12-30:
Which specific MMAL component are you thinking of?

Most of them will never drop buffers. They will hold input buffers until they have sufficient output buffers to decode into.

The easiest way of tracking input to output buffers is to set the timestamp on the input buffer, as provided the input buffer contains precisely one buffer (and ideally has the MMAL_BUFFER_HEADER_FLAG_FRAME_END flag set), then the corresponding output buffer will also have the same timestamp applied.

The one exception I can think of is video_render which will return input buffers once they are not on the display. If you submit a second input buffer before the display VSYNC then the first should be returned as soon as the second is submitted. A buffer on the display will be held until after the VSYNC which results in it being removed.

linhuifj, 2020-12-31:
I am using "vc.ril.image_decode", and I found that it takes 100ms for the decoder to decode a 3840x2160 image.  I can send the input at 20fps, but only get 10fps from the output, and the system memory consumption does not increase, so I think half of the packets are dropped.

I also try to set buffer->pts and buffer->dts when I input the frame, but the output buffer's pts and dts is either 0 or some fixed large number.