raspberrypi/userland #675: All time-related libc function calls in Docker images based on alpine > 3.13.1 fail

lxgr, 2021-02-08:
**Describe the bug**
libseccomp2 is incompatible with newer alpine-based Docker images and causes all time-related libc functions to fail.

**To reproduce**
```
~ docker run -it --rm alpine:latest
/ # date
```

**Expected behaviour**
```
~ docker run -it --rm alpine:3.12.3
/ # date
Sun Feb  7 18:02:12 UTC 2021
```

**Actual behaviour**
```
~ docker run -it --rm alpine:latest
/ # date
Sun Jan  0 00:100:4174038  1900
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage5
```
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?
`Linux <redacted> 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
See also https://github.com/alpinelinux/docker-alpine/issues/141#issuecomment-775166894 and https://wiki.alpinelinux.org/wiki/Release_Notes_for_Alpine_3.13.0#musl_1.2

jamesrr39, 2021-02-12:
I also ran into the same problem (also on armv7l architecture). I found this answer on stack overflow: https://askubuntu.com/a/1264921, which solved it for me. However instead of download + dpkg, I got the new libseccomp2 through apt:

```
apt update && apt install libseccomp2
```

I also tried running the container with `--privileged` out of curiosity, before updating libseccomp2. It worked, but of course upgrading libseccomp2 is much more preferable solution.

pelwell, 2021-02-12:
This was never going to be the correct place to report this issue, but I'm happy you've found a solution. Even though closed, the issue and answer will still be searchable.

closed 2021-02-12 by pelwell