raspberrypi/userland #676: Is there any plans to include OpenGL ES 3.1 in this repository?

kehribar, 2021-02-14:
Hi everyone,

Is there any plans to update the OpenGL ES libraries in this repository?

"rpi-userland" package (https://github.com/buildroot/buildroot/blob/master/package/rpi-userland/rpi-userland.mk) in the buildroot uses this repository as is. If someone wants to use mmal library for example, this package is a must to install. 

But the problem (?) is that libegl and libgles libraries in this repo are only 2.0 and doesn't support "compute shader" functionality which comes with 3.1 Also when I try to install mesa3d package for the 3.1 library, build system complains about "two different libegl, libgles libraries" and wants me to choose between "rpi-userland" and "mesa3d" packages.

This may not be a raspberrypi/userland problem per se, but I wanted to check whether is there any plans to update the OpenGL libraries in this repo anyway.

Thanks for everything
ihsan.

tombsar, 2021-02-14:
I believe this repo is fixed at GLESv2 to match the version supported by the binary Broadcom driver, which will never be updated.

kehribar, 2021-02-14:
Let me answer my question with the following post:

> There is no legacy / firmware GL driver on the Pi4. The V3D block has been upgraded (from VideoCore 4 to VideoCore 6) and the firmware has no driver for it.
Legacy mode gives you a DispmanX based framebuffer (instead of DRM/KMS) but no GL/GLES at all on Pi4, and that isn't going to change.

Source: https://www.raspberrypi.org/forums/viewtopic.php?t=12581#p1779930

So the GL version in this repository is not just old but not even compatible with the PI4 GPU version. One must use the mesa library to facilitate the GPU. That's the only option.

I'm closing the issue. 

Thanks again.

closed 2021-02-14 by kehribar