raspberrypi/userland #677: Enter 10 and get 16 for free

rrpegasus, 2021-02-22:
**Is this the right place for my bug report?**
free42-nologo/stable,now 1.4.77-1.2 armhf

**Describe the bug**
Humongus number entry bug.

**To reproduce**
Start /usr/bin/free42bin. Enter 10 and hit [Enter].

**Expected behaviour**
Have 10 (twice) on the stack as entered.

**Actual behaviour**
You get 16 (twice) on the stack.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 400
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2020-08-20
* Which firmware version (`vcgencmd version`)? Jan 27 2021 22:19:57 (c) 2012 Broadcom version 99d9a48302e4553cff3688692bb7e9ac760a03fa
* Which kernel version (`uname -a`)? Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

**Logs**
N/A

**Additional context**
N/A

pelwell, 2021-02-22:
That's not our code or something we support. Try the maintainers of free42.

closed 2021-02-22 by pelwell

HinTak, 2021-02-23:
Correct place to report this would be https://github.com/thomasokken/free42