raspberrypi/userland #682: Assertion failure in mmal_vc_client.c:204

kiran99nv, 2021-03-17:
**Describe the bug**
*Setup details*
1. Used 64bit version of Raspberry Pi OS from https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/
2. Used 64bit version Of Userland Libraries [Commit ID "7d3c6b9f4c3ddeecefdeb2b882bada74a235249b"]
3. Used OmniVision OV7251 (640 x 480) camera, which uses Userland MMAL Libraries
 
*Observation*
Using a simply camera application, which just stream's the live data frames captured. When the Camera is running continuous for more than 8hr, observed **_assertion failure:/home/pi/userland/interface/mmal/vc/mmal_vc_client.c:204:mmal_vc_lookup_client_context():client_context_pool.contexts[CLIENT_CONTEXT_MAGIC_MASK(index)].inuse_**  and application is crashed. 

**To reproduce**

- Use 64bit Raspberry PI OS and build 64 bit Userland libraries on Raspberry Pi 4 Model B Rev 1.2
- Connect camera to RPI4 board and run any simple Camera Application which uses MMAL libraries 
- Let the camera application run for more than 8hrs continuously.
- Note:- sometimes crash may occur within 8hrs or may occur in 8-48hrs,   

**Expected behavior**
In Longrun Camera Application will crash due to **_assertion failure:/home/pi/userland/interface/mmal/vc/mmal_vc_client.c:204:mmal_vc_lookup_client_context():client_context_pool.contexts[CLIENT_CONTEXT_MAGIC_MASK(index)].inuse_**

**Actual behaviour**
Camera Application should not crash 

**System**
System Information
------------------
Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

**cat /etc/rpi-issue**
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4

**vcgencmd version**
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start_x)

**uname -a**
Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

**Logs**
Please find the Application Logs with GDB debug logs below
 ./picam_test 
i2c: write: Remote I/O error
i2c: write: Remote I/O error
Serial Number: 01230e59686d9786ee
Width=640, Height=480, Size=384000
camera found: OmniVision OV7251 (640 x 480)
-----------------------------
 q - Quit
 e - exposure UP
 E - exposure DOWN
 g - gain UP
 G - gain DOWN
 p - up FPS
 P - down FPS
 f - flash ON
 F - flash OFF
 o - set continuous flash mode
 O - unset continuous flash mode
 a - set gamma 1.0
 b - set gamma 1.5
 c - set gamma 2.2
 t - read sensor temperature
 l - read laser voltage
 m - show min/max
 M - hide min/max
 u - cumulation mode (compare-max)
 U - cumulation mode off
[... 721 lines cut ...]
[New LWP 26814]
[New LWP 26819]
[New LWP 26813]
[New LWP 26811]
[New LWP 26812]
[New LWP 26806]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Core was generated by `./picam_test'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7fa33a5f90 (LWP 26808))]
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000007fb2e9f8e8 in __GI_abort () at abort.c:79
#2  0x0000007fb32d4c74 in vcos_pthreads_logging_assert (file=0x7fb32b3938 "/home/pi/userland/interface/mmal/vc/mmal_vc_client.c", func=0x7fb32b4380 <__func__.10501> "mmal_vc_lookup_client_context", line=204, fmt=0x7fb32b3930 "%s")
    at /home/pi/userland/interface/vcos/pthreads/vcos_pthreads.c:560
#3  0x0000007fb32ac2f4 in mmal_vc_lookup_client_context (index=-19136507) at /home/pi/userland/interface/mmal/vc/mmal_vc_client.c:204
#4  0x0000007fb32ad2b8 in mmal_vc_vchiq_callback (reason=VCHIQ_MESSAGE_AVAILABLE, vchiq_header=0x7f9c006cc8, service=263, context=0x0) at /home/pi/userland/interface/mmal/vc/mmal_vc_client.c:540
#5  0x0000007fab67866c in completion_thread (arg=0x7fab68b1f8 <vchiq_instance>) at /home/pi/userland/interface/vchiq_arm/vchiq_lib.c:1554
#6  0x0000007fb32d3f70 in vcos_thread_entry (arg=0x7fab68b208 <vchiq_instance+16>) at /home/pi/userland/interface/vcos/pthreads/vcos_pthreads.c:144
#7  0x0000007fb2e437e4 in start_thread (arg=0x7fd925a74f) at pthread_create.c:486
#8  0x0000007fb2f4eadc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
(gdb) quit

kiran99nv, 2021-03-17:
[image]
[image]

6by9, 2021-03-17:
./picam_test is not an application written or supported by Raspberry Pi.

There is an OV7251 driver and device tree overlay in the kernel. The recommendation would be to use that.

MMAL is not supported in 64 bit OSes at present. It's not built when ARM64 is defined - https://github.com/raspberrypi/userland/blob/master/CMakeLists.txt#L11
The fact that you have chosen an old revision of the userland repo (https://github.com/raspberrypi/userland/commit/7d3c6b9f4c3ddeecefdeb2b882bada74a235249b from 11 Dec 2019) is not our problem. 64bit support will be investigated again as and when time allows.