raspberrypi/userland #685: dtoverlay: Failed to find overlays directory

lokeb, 2021-03-21:
I am trying to run `dtoverlay` on Debian-arm64 after compiling userland from source. `buildme` installed the compiled binaries into `/opt/vc`. I added lib and bin to proper paths.

On running, `dtoverlay` reports `Failed to find overlays directory`. 

**System**
* Raspberry Pi 4 
* Debian arm64
* Jun  1 2020 13:23:40 
  Copyright (c) 2012 Broadcom
   version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start)

* Linux 5.9.0-0.bpo.5-arm64 #1 SMP Debian 5.9.15-1~bpo10+1 (2020-12-31) aarch64 GNU/Linux

Additionally, are there any Raspios repositories that I could use directly with Debian, because all `rpi-` commands are absent in my system?

pelwell, 2021-03-21:
Does /boot/overlays exist? If not, where are the overlays in Debian?

HinTak, 2021-03-21:
FWIW, ubuntu uses `/boot/firmware/overlays`. No idea about debian-arm64 .

pelwell, 2021-03-21:
The app looks in /boot/overlays and /flash/overlays. You can override that with the `-d` option. It is also possible to build the app specifying CPP macro values `FORCE_BOOT_DIR` and `OTHER_BOOT_DIR`.

tmm1, 2021-10-14:
> It is also possible to build the app specifying CPP macro values `FORCE_BOOT_DIR` and `OTHER_BOOT_DIR`.

How do I ask cmake to set a value for `OTHER_BOOT_DIR`?