raspberrypi/userland #686: raspistill hangs in a tight loop on the RPi4 after about an hour

iamsrp, 2021-03-22:
**Describe the bug**
When run in a tight loop, raspistill will hang on an RPi4 after about an hour. At 6s per iteration this is  . . . an exercise left to the reader. This is reproduced with both the regular camera and the HD one. I've not been able to reproduce this on a Pi Zero W or the original Pi1B.

**To reproduce**
This consistently works for me on two different RPi4s:
```
while true; do date; raspistill -n -o /tmp/foo.jpg; sync; done
```

**Expected behaviour**
Not to hang.

**Actual behaviour**
Hangs after about an hour or so.

**System**
This is without the camera attached to the Pi4 -- it's busy trying to reproduce on a PiZeroW.

```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2

Linux widepi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
Revision	: c03111
Serial		: 10000000e284549f
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Feb 25 2021 12:11:39 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       61105308 4309624  54280604   8% /
devtmpfs         1801156       0   1801156   0% /dev
tmpfs            1933252       0   1933252   0% /dev/shm
tmpfs            1933252    8608   1924644   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1933252       0   1933252   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48783    209312  19% /boot
tmpfs             386648       0    386648   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20210310
[... 967 lines cut ...]
[    6.914250] uart-pl011 fe201000.serial: no DMA platform data
[    7.020407] random: crng init done
[    7.020430] random: 7 urandom warning(s) missed due to ratelimiting
[    7.069868] 8021q: 802.1Q VLAN Support v1.8
[    7.264061] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.392683] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    7.612521] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.612771] bcmgenet fd580000.ethernet eth0: Link is Down
[   10.713808] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   10.713864] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.240987] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.745222] Bluetooth: Core ver 2.22
[   13.745327] NET: Registered protocol family 31
[   13.745344] Bluetooth: HCI device and connection manager initialized
[   13.745374] Bluetooth: HCI socket layer initialized
[   13.745396] Bluetooth: L2CAP socket layer initialized
[   13.745434] Bluetooth: SCO socket layer initialized
[   13.761375] Bluetooth: HCI UART driver ver 2.3
[   13.761397] Bluetooth: HCI UART protocol H4 registered
[   13.761525] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.762420] Bluetooth: HCI UART protocol Broadcom registered
[   13.934008] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.934017] Bluetooth: BNEP filters: protocol multicast
[   13.934030] Bluetooth: BNEP socket layer initialized
```

iamsrp, 2021-03-24:
I've been unable to reproduce this on the Pi1 or PiZero. Not had access to other models to test with.

hastou, 2021-05-20:
I have the same issue with tens of raspberry pi 4. Here are the logs of one of them with start_debug=1 in config.txt, and commands `sudo vcdbg log assert` and `sudo vcdbg log msg` after raspistill hung. The issue is the same on others Raspberry with different kernel  and firmware.

### Hardware
Raspberry Pi 4 Model B Rev 1.1

### Firmware version
```
Apr 30 2021 13:46:22 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start_db)
```

### Kernel version
`Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux`

### Rpi Issue
```
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
```

### vcdbg log assert
```
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
[... 1931 lines cut ...]
2894503.052: Using RPI AWB
2901167.339: Using RPI AWB
2907834.242: Using RPI AWB
2914502.298: Using RPI AWB
2921165.074: Using RPI AWB
2927830.187: Using RPI AWB
2934497.961: Using RPI AWB
2941170.844: Using RPI AWB
2947847.061: Using RPI AWB
2954510.740: Using RPI AWB
2961175.411: Using RPI AWB
2967841.511: Using RPI AWB
2974508.070: Using RPI AWB
2981182.149: Using RPI AWB
2987849.048: Using RPI AWB
2994517.688: Using RPI AWB
3001195.634: Using RPI AWB
3007862.194: Using RPI AWB
3014525.187: Using RPI AWB
3021194.360: Using RPI AWB
3027861.657: Using RPI AWB
3034530.692: Using RPI AWB
3041195.197: Using RPI AWB
3048753.700: mmal: fail_destroy: Timed out waiting to destroy ril.camera
```

hastou, 2021-05-21:
@6by9 This could be a duplicate of https://github.com/raspberrypi/firmware/issues/1574.

iamsrp, 2021-08-07:
Looks like 1574 has been closed, possibly with a workaround. Has anyone here been able to reproduce the issue in this ticket?

dfontenot, 2021-09-29:
I'm also running into this problem with my Pi4 (2GB). Was able to use a Pi camera v1.2 just fine. Switched to a OV5647 5MP camera, got ENOMEM errors from mmal, upped the gpu mem from 128 to 256. Now raspistill just hangs when I run it.