raspberrypi/userland #695: CM4 host

gregoryjaooi, 2021-06-23:
Will the codes include CM4 (x14) and PI400 (x13) model type detection ?
Or may be there should be a better work around.
Some users are using CM4 and the source codes are using below function call.
As such, the same application has error/issues running on CM4 or PI400 while
on PI4 (x11) there are no issues at all.

/* Returns the type of the Pi being used
*/
int bcm_host_is_model_pi4(void)
{
   return bcm_host_get_model_type() == 0x11 ? 1 : 0;
}

pelwell, 2021-06-23:
@popcornmix may have some thoughts on the matter, but as you say I think this function makes more sense as a way of detecting any member of the Pi 4 family, i.e. 4B, 400 and CM4. This is equivalent to checking if the processor is a BCM2711, so I think it should be implemented in that way. While looking at this I also noticed that the list of processor types include BCM2838 - our guess at the Broadcom name for what turned out to be called BCM2711. `BCM_HOST_PROCESSOR_BCM2711` should be included as an alias.

pelwell, 2021-06-23:
See https://github.com/raspberrypi/userland/pull/696.

gregoryjaooi, 2021-06-25:
This issue can be closed as CM4
is running very well now after raspi-update.
Thanks.

closed 2021-06-25 by popcornmix