raspberrypi/userland #701: Non-open source Non-compatible copyrighted and patented proprietary VNC software included in default distro

gitcnd, 2021-07-12:
**Describe the bug**
VNC(R) Server 6.7.2 (r42622) ARMv6 (May 13 2020 19:33:04), which is Copyright (C) 2002-2020 RealVNC Ltd, is part of this distro.
It is not open source, and it does not work with open source products - it causes "No Matching Security Types" errors, because ReadVNC patented their product.

**To reproduce**
Enable VNC and attempt to connect with an open-source VNC client

**Expected behaviour**
Works.

**Actual behaviour**
"No Matching Security Types" error

**System**
$ raspinfo
System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

Linux raspberrypi 5.10.17+ #1421 Thu May 27 13:58:02 BST 2021 armv6l GNU/Linux
Revision        : 9000c1
Serial          : 000000009fd70633
Model           : Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

May 27 2021 14:05:31 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7120120 4359396   2418756  65% /
devtmpfs          154744       0    154744   0% /dev
tmpfs             187724   12516    175208   7% /dev/shm
tmpfs             187724    2864    184860   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             187724       0    187724   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   49690    208406  20% /boot
tmpfs              37544       0     37544   0% /run/user/0
tmpfs              37544       4     37540   1% /run/user/1000

Filename                                Type            Size            Used            Priority
/var/swap                               file            102396          101820          -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20210706
raspberrypi-sys-mods:
  Installed: 20210706
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
[... 516 lines cut ...]
[  512.739391] w1_master_driver w1_bus_master1: Attaching one wire slave 00.100000000000 crc 9d
[  512.790971] w1_master_driver w1_bus_master1: Family 0 for 00.100000000000.9d is not registered.
[  555.237986] w1_master_driver w1_bus_master1: Attaching one wire slave 00.900000000000 crc 11
[  555.305533] w1_master_driver w1_bus_master1: Family 0 for 00.900000000000.11 is not registered.
[  634.540178] w1_master_driver w1_bus_master1: Attaching one wire slave 00.500000000000 crc db
[  634.596204] w1_master_driver w1_bus_master1: Family 0 for 00.500000000000.db is not registered.
[  689.902350] w1_master_driver w1_bus_master1: Attaching one wire slave 00.d00000000000 crc 57
[  689.988275] w1_master_driver w1_bus_master1: Family 0 for 00.d00000000000.57 is not registered.
[  717.893228] w1_master_driver w1_bus_master1: Attaching one wire slave 00.300000000000 crc be
[  717.942646] w1_master_driver w1_bus_master1: Family 0 for 00.300000000000.be is not registered.
[  769.923834] w1_master_driver w1_bus_master1: Attaching one wire slave 00.b00000000000 crc 32
[  769.981108] w1_master_driver w1_bus_master1: Family 0 for 00.b00000000000.32 is not registered.
[  811.434431] w1_master_driver w1_bus_master1: Attaching one wire slave 00.700000000000 crc f8
[  811.485604] w1_master_driver w1_bus_master1: Family 0 for 00.700000000000.f8 is not registered.
[  866.764093] w1_master_driver w1_bus_master1: Attaching one wire slave 00.f00000000000 crc 74
[  866.819789] w1_master_driver w1_bus_master1: Family 0 for 00.f00000000000.74 is not registered.
[  932.006326] w1_master_driver w1_bus_master1: Attaching one wire slave 00.080000000000 crc c2
[  932.087991] w1_master_driver w1_bus_master1: Family 0 for 00.080000000000.c2 is not registered.
[  989.058706] w1_master_driver w1_bus_master1: Attaching one wire slave 00.880000000000 crc 4e
[  989.111571] w1_master_driver w1_bus_master1: Family 0 for 00.880000000000.4e is not registered.
[ 1028.646096] w1_master_driver w1_bus_master1: Attaching one wire slave 00.480000000000 crc 84
[ 1028.697859] w1_master_driver w1_bus_master1: Family 0 for 00.480000000000.84 is not registered.

**Additional context**
Perhaps TigerVNC would be a better product - it's open and already default on other distros.

popcornmix, 2021-07-12:
Sorry - this repo is for code in the userland tree (primarily for communicating with GPU).
VNC is not part of this repo.
https://github.com/RPi-Distro/repo/issues/ may be a better place.

closed 2021-07-12 by popcornmix

pelwell, 2021-07-12:
RealVNC is included because RealVNC Connect provides cross-internet access - a valuable service for many users. For others, `sudo apt install tigervncserver` isn't a large hurdle.

JamesH65, 2021-07-12:
RealVNC is also has custom Pi specific code for acceleration, so we won't be removing it by default. It's easy enough, as @pelwell says, to install Tiger and if you want to, remove RealVNC yourself.