raspberrypi/userland #709: Documentation for the `pwm_speedup` command in vcgencmd

seamusdemora, 2021-11-02:
`vcgencmd commands` lists the command `pwm_speedup`. It also provides [this link](https://www.raspberrypi.org/documentation/computers/os.html#vcgencmd) to "further documentation. I don't see any documentation for the `pwm_speedup` command at this URL. Does any documentation exist? If so, where can I find it?

pelwell, 2021-11-02:
I'd not come across it before, but from reading the code it simply acts as a multiplier for the audio PWM clock. The commit dates back to 2013 and the message says `Add pwm_speedup hack for benchmarking`.

Nothing to see here - move along.

closed 2021-11-02 by pelwell

seamusdemora, 2021-11-15:
@pelwell 
> Nothing to see here - move along.

OK - I appreciate the feedback, but your answer also suggests that you feel this is "not your problem"?? Or am I reading something into it? I was under the impression this code was unique to RPi hardware, so apologies if I've brought someone else's oversight to your doorstep.

popcornmix, 2021-11-17:
Some gencmd's (and other interfaces) were purely added for RPi engineers to debug things.
Use of them is likely to make the system behave in unexpected ways.
Documenting these interfaces gives the impression they are supported and is likely to generate more confusion.

seamusdemora, 2021-11-22:
> Documenting these interfaces gives the impression they are supported and is likely to generate more confusion.

I suppose one idea might be to remove any reference to unsupported gencmd's. That has the obvious benefit of saving me time looking for something that doesn't exist - and for you having to answer a question that's pursued doggedly.