raspberrypi/userland #710: host_applications/linux/apps/hello_pi/hello_fft/mailbox.h: Remove declaration of unsigned get_version?

nubok, 2021-11-08:
**Describe the bug**

The file `mailbox.h` of the hello_fft application

https://github.com/raspberrypi/userland/blob/3fd8527eefd8790b4e8393458efc5f94eb21a615/host_applications/linux/apps/hello_pi/hello_fft/mailbox.h 

contains a declaration

`unsigned get_version(int file_desc);`

This declaration is used nowhere in the code. In particular

https://github.com/raspberrypi/userland/blob/3fd8527eefd8790b4e8393458efc5f94eb21a615/host_applications/linux/apps/hello_pi/hello_fft/mailbox.c

does not contain an implementation of this function.

**Expected behaviour**

I would expect that either

1. `mailbox.c` contains an implementation of `get_version` or
2. the declaration of `get_version` is removed from `mailbox.h` since no implementation of it exits.

If it is decided to choose 1., it would make sense that `get_version` is used in hello_fft (https://github.com/raspberrypi/userland/tree/3fd8527eefd8790b4e8393458efc5f94eb21a615/host_applications/linux/apps/hello_pi/hello_fft).

**Actual behaviour**

See section "Describe the bug".

popcornmix, 2021-11-08:
Removed the unused declaration from internal repo. Will appear in userland next firmware build.

popcornmix, 2021-11-08:
Userland updated.