raspberrypi/userland #722: ./buildme  doesn't work

hamedmh, 2022-03-18:
Hi,
I got many errors related to missing files that should be found in folder "interface/vcsm/", when building vcsm and user-vcsm.c.o

Is there a solution for this issue? I use Raspberry Pi 4B.

Thanks!

Leo3418, 2022-03-18:
It would have been more helpful if you could have pasted the command you had used and the full output of that command...

6by9, 2022-03-18:
And details of the system you are running it on (32 or 64bit, and Raspberry Pi OS release).

I've just run it on a 32bit Raspberry Pi OS, and it's built fine (although there are more warnings than ideal).

The build information for host_applications/linux/libs/sm/ does install the user-vcsm.h file in include/interface/vcsm under the target path.

hamedmh, 2022-03-18:
Thank you!
I've installed the arm64 version.
Should I install a 32bit system?

The list of warnings and notes is too long. Here is what was left in the terminal when running  ./buildme

_client.c:396:7: note: in expansion of macro ‘vcos_log_trace’
  396 |       vcos_log_trace("Send make current %d[%d %s%s] %d[%d %d%s]",
      |       ^~~~~~~~~~~~~~
make[2]: *** [interface/khronos/CMakeFiles/EGL_static.dir/build.make:368: interface/khronos/CMakeFiles/EGL_static.dir/common/khrn_client.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1573: interface/khronos/CMakeFiles/EGL_static.dir/all] Error 2
make[2]: *** [interface/khronos/CMakeFiles/khrn_static.dir/build.make:368: interface/khronos/CMakeFiles/khrn_static.dir/common/khrn_client.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  5%] Building ASM object interface/khronos/CMakeFiles/khrn_static.dir/common/khrn_int_hash_asm.s.o
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s: Assembler messages:
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:36: Error: unknown architecture `armv6'

/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:37: Error: unknown pseudo-op: `.object_arch'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:38: Error: unknown pseudo-op: `.arm'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:104: Warning: unknown register 'a1' -- .req ignored
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:105: Warning: unknown register 'a2' -- .req ignored
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:106: Warning: unknown register 'a3' -- .req ignored
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:107: Warning: unknown register 'a4' -- .req ignored
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:110: Warning: unknown register 'ip' -- .req ignored
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:113: Error: operand 1 must be an integer register -- `ldr BB,=0xDEADBEEF'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:114: Error: unknown mnemonic `push' -- `push {CC,DAT0,lr}'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:115: Error: operand 1 must be an integer or stack pointer register -- `add AA,AA,N,lsl#2'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:116: Error: operand 1 must be an integer or stack pointer register -- `add AA,AA,BB'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:117: Error: operand 1 must be an integer register -- `mov BB,AA'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:118: Error: invalid use of vector register at operand 1 -- `mov CC,AA'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:121: Error: operand 1 must be an integer or stack pointer register -- `cmp N,#(prefetch_distance+2)*32/4'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:128: Error: operand 1 must be an integer or stack pointer register -- `sub N,N,#(prefetch_distance+2)*32/4'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:129: Error: invalid use of vector register at operand 1 -- `bic DAT0,S,#31'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:134: Error: unknown mnemonic `pld' -- `pld [DAT0,#OFFSET]'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:134: Error: unknown mnemonic `pld' -- `pld [DAT0,#OFFSET]'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:134: Error: unknown mnemonic `pld' -- `pld [DAT0,#OFFSET]'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:135: Error: operand 1 must be an integer or stack pointer register -- `and DAT1,S,#0x1C'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:136: Error: operand 1 must be an integer or stack pointer register -- `cmp DAT1,#0x0C'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:141: Error: unknown mnemonic `ldmia' -- `ldmia S!,{DAT0,DAT1,DAT2}'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:142: Error: operand 1 must be an integer or stack pointer register -- `sub N,N,#12/4'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:143: Error: operand 1 must be an integer or stack pointer register -- `add AA,AA,DAT0'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:144: Error: operand 1 must be an integer or stack pointer register -- `add BB,BB,DAT1'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:145: Error: invalid use of vector register at operand 1 -- `add CC,CC,DAT2'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: operand 1 must be an integer or stack pointer register -- `sub AA,AA,CC'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: operand 1 must be an integer or stack pointer register -- `eor AA,AA,CC,ror#32-4'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: invalid use of vector register at operand 1 -- `add CC,CC,BB'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: operand 1 must be an integer or stack pointer register -- `sub BB,BB,AA'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: operand 1 must be an integer or stack pointer register -- `eor BB,BB,AA,ror#32-6'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:146: Error: operand 1 must be an integer or stack pointer register -- `add AA,AA,CC'
[... cut ...]
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:221: Error: unknown register alias 'BB'
/home/pi/userland/interface/khronos/common/khrn_int_hash_asm.s:224: Error: unknown register alias 'DAT1'
make[2]: *** [interface/khronos/CMakeFiles/khrn_client.dir/build.make:121: interface/khronos/CMakeFiles/khrn_client.dir/common/khrn_int_hash_asm.s.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1546: interface/khronos/CMakeFiles/khrn_client.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
~/userland

6by9, 2022-03-18:
From the README of this repo (and why providing all the details is critical):

> Whilst 64-bit userspace is not officially supported, some of the libraries will work for it. To cross compile, install gcc-aarch64-linux-gnu and g++-aarch64-linux-gnu first. For both native and cross compiles, add the option ```--aarch64``` to the buildme command.

./buildme --aarch64```

hamedmh, 2022-03-18:
Great!  Thanks!

pelwell, 2022-03-18:
Please close the Issue when you have successfully completed the built (noting the proviso that not everything works in a 64-bit userspace).

hamedmh, 2022-03-19:
Thanks for great help!

closed 2022-03-19 by hamedmh