raspberrypi/userland #727: vcgencmd display_power 0 does not turn off screen after updating Raspberry Pi OS

ismarslomic, 2022-08-27:
**Describe the bug**
After updating Raspberry Pi OS (bullseye) on Raspberry Pi 4 with `sudo apt update && sudo upgrade -y` yesterday, I'm no longer able to turn on/off my LCD screen (connected with HDMI) with command `vcgencmd display_power 0 / 1` and the screen stays on. It worked as expected prior to the update.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
`vcgencmd display_power 0` turns screen off and `vcgencmd display_power` shows current power state `0`

**Actual behaviour**
`vcgencmd display_power 0` does not turn screen off and `vcgencmd display_power` shows current power state `1`

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

Linux raspberrypi 5.15.56-v7l+ #1575 SMP Fri Jul 22 20:29:46 BST 2022 armv7l GNU/Linux
Revision	: c03111
Serial		: 100000003f786852
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=1, libcamera interfaces=0

Videocore information
---------------------

Aug  9 2022 13:45:36
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
df: /run/user/1000/doc: Operation not permitted
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122602552 12311332 105257440  11% /
devtmpfs         1774272        0   1774272   0% /dev
tmpfs            1939136        0   1939136   0% /dev/shm
tmpfs             775656     1140    774516   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    258095    51693    206403  21% /boot
tmpfs             387824       28    387796   1% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20220530
raspberrypi-sys-mods:
  Installed: 20220224
openbox:
  Installed: 3.6.1-9+rpt1+deb11u1
lxpanel:
  Installed: 0.10.1-2+rpt19
pcmanfm:
[... 671 lines cut ...]
[   19.632213] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.632239]  MAI: soc_pcm_open() failed (-19)
[   19.890424] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.890465]  MAI: soc_pcm_open() failed (-19)
[   19.891885] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.891922]  MAI: soc_pcm_open() failed (-19)
[   19.892831] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.892860]  MAI: soc_pcm_open() failed (-19)
[   19.904555] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.904596]  MAI: soc_pcm_open() failed (-19)
[   19.904968] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.904991]  MAI: soc_pcm_open() failed (-19)
[   19.905363] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.905395]  MAI: soc_pcm_open() failed (-19)
[   19.936318] Bluetooth: RFCOMM TTY layer initialized
[   19.936347] Bluetooth: RFCOMM socket layer initialized
[   19.936374] Bluetooth: RFCOMM ver 1.11
[   32.473291] cam-dummy-reg: disabling
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

JamesH65, 2022-08-30:
It sounds like you have started using the KMS driver, having moved from either the FKMS or the legacy graphics stack. That requires using the standard Linux command for turning displays on and off, not vcgencmd

ismarslomic, 2022-08-30:
If that is the case, then I did not do this switch intentionally. The only change I did, before things got broken, was to run `sudo apt update && sudo upgrade -y`. What standard Linux command would you recommend, as replacement for the `vcgencmd`? I use the LCD screen on my smart mirror, and I just need to be able to turn it on/off when needed, through CLI.

ismarslomic, 2022-09-08:
I just did a fresh reinstall of Raspberry Pi OS and installed the new updates on another rpi . Then I tested turning off the screen with `vcgencmd display_power 0`, bot same issue. Screen is still powered on! 

Can anyone try reproducing these on their rpi?

timg236, 2022-09-09:
See comment from JamesH65. "vcgencmd display_power 0" is only supported by the legacy FKMS display stack (vc4-fkms-v3d). The default is now KMS (vc4-fkms-v3d)

Use the xset dpms commands to control display power with KMS
https://www.computerhope.com/unix/uxset.htm

JamesH65, 2022-09-09:
See also this forum post. https://forums.raspberrypi.com/viewtopic.php?p=2036401&hilit=display_power#p2036401

ismarslomic, 2022-09-09:
Thanks for helping out! I managed to turn the screen off/on with `xset` and `xrandr`, sharing details below so other users can find it later.

```
# tell xset which display to use
export DISPLAY=:0

# xandr turn off
xrandr --output HDMI-1 --off

# xandr turn on
xrandr --output HDMI-1 --auto

# xset turn off
xset dpms force off

# xset turn on
xset dpms force on
```

Important note: Screen Blanking must be enabled if using `xset`, otherwise you will see error saying "server does not have extension for DPMS option", ref https://forums.raspberrypi.com/viewtopic.php?t=277548

closed 2022-09-09 by ismarslomic