raspberrypi/userland #731: Invalid command line option (-roi)

lmitcham, 2023-05-25:
I am using a Zero 2W with camera in legacy mode and successfully using raspivid, however, when I attempt to use -roi, I get this message:

me@pi:~ $ raspivid -t 0 –roi 0.25,0.25,0.47,0.36
Invalid command line option (–roi)

Yet this command works just fine:

raspivid -t 0 -md 3 -rot 180 -fps 30 -ss 10000 -o test.h264 -a 1032 -ae 120,0xFF,0x808000

pelwell, 2023-05-25:
If look at your command line carefully you might notice that it uses two different dash characters. The first (`-`) is a hyphen, which is correct. The second (`–`) is an _en dash_ (UTF character e2 80 93), which you sometimes see when text is copied from a webpage, Word document, etc. The application doesn't understand en dashes, so it complains.

lmitcham, 2023-05-25:
Awesome explanation ... thank you ... that was it.

closed 2023-05-25 by lmitcham